nvs-vocabs/L05 #6: NTR: Request for sediment suction samplers [BODCNVS-324]

louatbodc, 2019-12-10:
## Required
### Term name (PrefLabel)
sediment suction samplers
### Definition
Devices that collect samples of surface sediment using suction. The mechanism of suction can be accomplished by either vacuum, by pressure difference between the air inside the sampler and surrounding water, by pumping water directly into the sampler or by under pressure air to elevate the sediment inside the sampler.
### Sources/references
[Please enter a link to a reference source]

## Optional
### Synonym or acronym (AltLabel)
[Only use if the new device category has known synonyms or acronyms]
### Mapping to external terminologies
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]
### ORCID
[Enter your ORCID if desired]

neil-ices-dk, 2019-12-10:
Sounds fine, one ambiguity in the definition around the use of 'surface' assuming this means bottom or sediment layer surface.

louatbodc, 2019-12-11:
Hi Neil

How does this sound?

Definition

Devices that collect samples from the sediment layer surface using suction. The mechanism of suction can be accomplished by either vacuum, by pressure difference between the air inside the sampler and surrounding water, by pumping water directly into the sampler or by under pressure air to elevate the sediment inside the sampler. Devices are typically diver- or remotely-operated.

neil-ices-dk, 2019-12-12:
Sounds good Louise 👍

louatbodc, 2019-12-19:
New term created. Please note - it may talk uptimes to 24 hours for the term to appear on the NVS.

sediment suction samplers
http://vocab.nerc.ac.uk/collection/L05/current/391/

closed 2019-12-19 by louatbodc