nvs-vocabs/L05 #7: NTR: Request for L05 Ship navigation systems

gwemon, 2020-05-13:
<H2> Required </H2><p><b>Term name (PrefLabel)</b><p>Ship navigation systems <p><b>Definition</b><p> Ship navigation systems provide continuous position and navigation safety information for mariners on-board marine vessels. These systems interface with other sensors on-board a vessel to provide and record location, heading, speed, and depth information.<p><H2> Optional </H2><p><b>Sources/references</b><p> http://www.ecdis-info.com/about_ecdis.html ; https://www.marineinsight.com/marine-navigation/what-is-electronic-chart-display-and-information-system-ecdis/<p> <b>Synonym or acronym (AltLabel)</b><p> ECDIS; ECS; Marine Navigation, Electronic Chart System; Electronic Chart Display and Information System<p>

gwemon, 2020-05-14:
External request logged in BODC systems as BODCNVS-501

louatbodc, 2020-06-08:
Dear requestee

Many thanks for your request. Ships Navigation Systems is quite a broad term encompassing many different types of navigational instruments. Can I ask - is the term you are requesting for the chart display as opposed to the navigational instruments that integrate with them? Alternatively, are you asking for a term to cover the integrated instrumentation? If so, we have many terms that classify such instrumentation such as multibeam echo sounders, positioning systems, platform orientation systems etc.

Kind regards
Lou

louatbodc, 2022-05-10:
Closing issue as no activity since 2020

closed 2022-05-10 by louatbodc