nvs-vocabs/L05 #11: Clarity for specifics on the assignment of L05 term to Underwater Vision Profilers

ssmosstee, 2020-06-29:
Hello all,

I am reaching out to gain clarity on the appropriate assignment of an L05 term to an Underwater Vision Profiler instrument. The instrument in question takes microscopic images of zooplankton and large phytoplankton, and has software that can further identify and categorize images through machine learning.

From my understanding, three L05 terms could possible be used to describe this instrument. These are:
SDN::L05::180 - [underwater cameras](http://vocab.nerc.ac.uk/collection/L05/current/180/)
SDN::L05::21 - [plankton recorders](http://vocab.nerc.ac.uk/collection/L05/current/21/)
SDN::L05::352 - [plankton counters](http://vocab.nerc.ac.uk/collection/L05/current/352/)

Currently I am learning more toward one of either term 180 or 352. I believe term 21 might be referring to another type of device that skims across the water surface and uses other optical techniques to capture plankton data.

Term 180 could fit because the UVP is a sort of underwater camera, but so could term 352 as the instrument's purpose is to count and sort plankton species.

Would anyone have insight into how to approach this?

gwemon, 2020-06-30:
Hello @ssmosstee, because the UVP's design is optimised to target plankton samples in situ, I would recommend using SDN::L05::352 - plankton counters. It is defined as "Instruments that categorise and count plankton in the water column." which seems to fit the UVP perfectly. Strangely enough we do not yet have any instrument models linked to this category in L22.

roy-lowry, 2020-06-30:
Agree 352 is the way to go. One thought. The instrument seems to categorise the plankton but not count them. Maybe broaden the definition of 352 to 'Instruments that categorise and/or count plankton in the water column.' ?

ssmosstee, 2020-06-30:
Thanks very much @gwemon, @roy-lowry for clearing things up on my end!

I had a similar discussion with my colleague at Ocean Networks Canada about the L22 mapping of the UVP and she might include a request with her next batch of suggestions. Our particular model is from Hydroptic and I think it is one of the more popular models - often featured in papers

I agree that perhaps the plankton counter term could be expanded to include a more encompassing group of instruments. For example, there are camera systems that take images of plankton for counting, categorization, and 2D/3D spatial reconstruction for volumetric measurements.

Thanks again. :)

gwemon, 2020-07-28:
@ssmosstee @roy-lowry In your views, is it sufficient to make the change to the description field or do we need to review the preferred label too? What could be a collective name for the instruments you mention above? Optical plankton analysers?

roy-lowry, 2020-07-28:
I would prefer not to narrow the semantics of an L05 preferred label so adding 'optical' isn't a good idea. As soon as we put optical in there somebody will come up with an acoustic plankton counter!!

I would probably leave the label as is or if a change is really felt to be necessary I'd obviously broaden the semantics such as 'plankton categorisers and counters'

danibodc, 2020-07-30:
Taking into account the above, we have updated the preferred label for 352: 
'plankton categorisers and counters'

And description:
'Instruments that categorise and/or count plankton, and/or enable 2D/3D spatial reconstruction of plankton in the water column.'

Many thanks!

closed 2020-07-30 by danibodc