nvs-vocabs/L05 #12: NTR: Request for L05 profiling sonars

nvsvocabs, 2020-11-19:
<H2> Required </H2><p><b>Term name (PrefLabel)</b><p>profiling sonars <p><b>Definition</b><p> High-frequency (>1 MHz), possibly multiple frequency, active sonar systems that provide high-speed cross-sectional profile of the target area. They are deployed on a surface survey vessel pointing the transducer down towards the target or mounted on an ROV/AUV/UUV. Applications of profiling sonars include cross-sectional profiling in pipeline and trench survey, bottom profiles, and surveying of underwater cave systems or manmade structures like road and rail bridge foundations. Examples of this type of system are Tritech Super SeaKing DFP Profiler and Imagenex 881A Profiling Sonar Head.<p><H2> Optional </H2><p><b>Sources/references</b><p> https://www.tritech.co.uk/uploaded_files/Knowledge%20Base%20-%20Profiling%20Sonars.pdfhttps://www.tritech.co.uk/uploaded_files/XMarketing0-Marketing-GeneralCommunicationsAdvertising-amp-Editorial2016Cave-Diving.pdfhttps://imagenex.com/products/881a-profiling<p> <b>ORCID</b><p> 0000-0002-7958-2191<p>

vpaba, 2021-02-03:
Hello,
Thanks for submitting this new term. As protocol, I have contacted the SeaVox community for feedback before proceeding. 
Regards,
Violetta

louatbodc, 2021-04-28:
An update to the term definition has been proposed to enable distinction from imaging sonars (http://vocab.nerc.ac.uk/collection/L05/current/MMIS/). The definition was re-proposed to SeaVox community who govern the L05 vocabulary collection:

Active sonar systems that provide a cross-sectional profile of a target area. These instruments typically emit a series of acoustic pulses in a beam pattern like a spot or pencil beam and the peak echo return is recorded. They provide a digitised version of echo returns as opposed to multicoloured images by imaging sonars. They are generally deployed on survey vessels pointing the transducer down towards the target. Applications of profiling sonars include cross-sectional profiling in pipeline and trench survey, bottom profiles, and surveying of underwater cave systems or manmade structures like road and rail bridge foundations. Examples of this type of system are Tritech Super SeaKing DFP Profiler and Imagenex 881A Profiling Sonar Head.

louatbodc, 2021-06-15:
Term now created:

(MPIS) profiling sonars

Please note it may take up to 24 hours for the new term to appear on our web services.

closed 2021-06-15 by louatbodc