nvs-vocabs/L05 #16: NTR: Request for L05 Automatic Identification Systems Receiver

nvsvocabs, 2022-09-02:
<H2> Required </H2><p><b>Term name (PrefLabel)</b><p>Automatic Identification Systems Receiver <p><b>Definition</b><p> Automatic Identification Systems (AIS) is a digital positional awareness system operating in the Very High Frequency (VHF) maritime band. It allows vessels and marine traffic services to automatically broadcast and receive static, dynamic, and voyage-related information to identify other ships and to monitor their movements, ensuring safe navigation through waterways.<p><H2> Optional </H2><p><b>Sources/references</b><p> https://e-navigation.canada.ca/topics/traffic/docs/ais/index-en<p> <b>ORCID</b><p> 0000-0002-3030-5177<p> <b>Synonym or acronym (AltLabel)</b><p> AIS Receiver<p>

roswri, 2022-11-21:
submitted to SeaVox governance group as:
Pref label: Automatic Identification Systems Receivers
Alt label: AIS
Definition: Digital positional awareness systems operating in the Very High Frequency (VHF) maritime band. They allow vessels and marine traffic services to automatically receive vessels’ identity, position, speed and course to identify other ships and to monitor their movements.
L21: http://vocab.nerc.ac.uk/collection/L21/current/ICAT06/

Feedback due by 22/11/2022

roswri, 2023-02-10:
Feedback received:

AIS transmitters are also put on buoys, platforms, SUV ‘s…
And they also capture status and destination..., I thought also cargo, but not sure of that.
Possible rephrase :
Digital positional awareness systems operating in the Very High Frequency (VHF) maritime band. They allow vessels and marine traffic services to automatically receive identity, position, speed, course, operational status and destination transmitted by other ships and platforms.

roswri, 2023-02-17:
New term created: POS27 - Automatic Identification System receivers [http://vocab.nerc.ac.uk/collection/L05/current/POS27/]

closed 2023-02-17 by roswri