nvs-vocabs/L05 #20: Update to existing term: Fish-finder echosounders

JordanAthertonNOC, 2022-11-29:
The acoustic device landscape has changed since this term was created - the term echosounder is applied to devices which have vertical directionality, whereas more contemporary fish finding devices also employ horizontal directionality in operation. Sonars can have vertical _or_ horizontal directionality, so all echosounders are a type of sonar, but a sonar is _not_ a type of echosounder.

In order to capture fish finding devices that have horizontal directionality we are proposing to amend the preflabel for the existing L05 term from `Fish-finder echosounders` to `Fish-finder acoustic instruments`, and enhancing the description to include the terms `sonars` and `echosounders` to aid discoverability. 

Seeking input from @roy-lowry before proceeding.

roy-lowry, 2022-11-30:
No restriction to vertical beams through the use of the word 'echosounder' was intended when this was set up. Many multi-beam echosounders - the instruments with which I was most familiar - have beams set in multiple directions so a directional semantic never entered my mind. If others do see a directional association then I've no problem with the changes proposed, although 'Acoustic fish finders' might be a little more elegant. 
I have encountered the term 'fish-finder echosounder' in common usage so it might be an idea to have that phrase in the definition (something like 'Includes fish-finder echosounders') so any searches for it still work.
I think the protocol of running changes to L05 past the SeaDataNet tech mailing list is still in force, although I've never noticed much in the way of responses these days.  Check with Gwen.

JordanAthertonNOC, 2022-11-30:
Present suggestion is to amend the preflabel from `fish-finder echosounders` to **either** `acoustic fish finders` **or** `fish-finder acoustic instruments`, in addition to amending the description to include the terms `sonars` and `fish-finder echosounders` to ensure discoverability. 

Awaiting further input from any interested parties in the wider community.

gavinmacaulay, 2022-12-05:
Some comments from a user of echosounders for fisheries research: 

1. I agree with @roy-lowry that ``echosounder``, as commonly used today, doesn't imply vertical directionality - where clarity is needed the directionality is often included (e.g., vertically-oriented echosounder, horizontally-oriented echosounder)

2. ``acoustic instruments`` is a very general label (more general than ``sonar``) that is too broad in this case.

3. The ``fish-finder`` label is very much associated with recreational fishing and it not commonly used in commercial fishing or fisheries research. But I see that ``fish-finder`` is used to distinguish from other types of echosounder (e.g., single- and multi-beam echosounders) - ``biological echosounder`` would be more a more accurate label but it's not a term that I've seen used at all and ``fish-finder`` at least has some historical meaning.

5. Not having been aware of this vocabulary until now, I looked at the definition and suggest something like the following to better reflect the use of fish-finder echosounders for more than just detecting fish shoals: ``Sonar instruments primarily designed to detect aquatic organisms through directed transmission and reception of acoustic signals. They often utilise multiple frequencies.``

In summary, I support retaining ``fish-finder echosounder`` and modifying the definition as above.

JordanAthertonNOC, 2023-01-03:
Hi @gavinmacaulay, thank you for your valuable input, we are happy to take forward your suggestion - the description for FFES has been amended accordingly and this update should be live after ~07:00 GMT tomorrow.

closed 2023-01-03 by JordanAthertonNOC