nvs-vocabs/L05 #22: NTR: Request for L05 telecommunication_facility

nvsvocabs, 2023-03-05:
<H2> Required </H2><p><b>Term name (PrefLabel)</b><p>telecommunication_facility <p><b>Definition</b><p> Entity or part of entity that enables telecommunication or can be utilized for telecommunication purposes  (e.g. via satellite internet constellation, by cell phones, optical fiber, or alike).<p><H2> Optional </H2><p><b>ORCID</b><p> 0000-0003-0367-6850<p>

nanselm, 2023-04-04:
hi, 

what's the schedule on issues?  this (and another one) is untouched for several weeks.

gwemon, 2023-04-04:
Sorry @nanselm this might have been missed on my watch. It's now logged in our systems. We will be back to you if required.

nanselm, 2023-04-04:
no worries, thanks for notifying. we are curious what is the general workflow (from here on). we (pangaea, marum, awi, +x) are planning for more. is this the preferred way? any obstacles or things to care about??

gwemon, 2023-04-04:
For just a few terms, github requests to the specific repo are fine but if you envisage a large volume of requests then it would be best if we talk it through in a quick meeting. Also please be aware that each term proposed need to fit the existing coverage of the existing vocabulary and requires careful consideration by the governance team. Some are more straightforward than others. I suspect that the term you are proposing here might require discussion. We have a team of people dedicated to managing instruments and platforms vocabularies. In order to provide more info could you give us some examples of the kind of instrument models you would see fitting in this category? Also would "telecomnunication devices" or "telecommunication equipment" be a better term?

nanselm, 2023-04-04:
The task group is on the way to harmonize the vocabularies in use in different systems and environments. This includes a lot discussion and everything that might be submitted has undergone some kind of review internally. At first we just care for devices and plan to have concise actions, such as development sprints. But this will take some time until we have agreed on a common workflow across the institutions. If I got our schedule right we intend to contact BODC anyway. 

Regarding this term:  we tried to have the definition as generic as possible. Typical devices would be starlink systems, oneweb, kongsberg marine broadcast systems, mobile gateways or smartphones used as modems in a measuring container (you know what I mean?). We omitted "devices" and chose "facility" because sometimes it is rather a bunch of things instead of a single device -- more a system. E.g. starlink has the antenna and a router box, the kongsberg mbr consist of an antenna, a massive 19" power unit and a computer. But we are open to discuss on this term suggestion. :D

gwemon, 2023-04-05:
Thank you @nanselm that's useful info. 
We already have a number of "systems" defined at the L05 level so this would probably be preferable to "facility" but let's see what the L05 management group thinks. @danibodc or @louatbodc or somebody else from the team will be in touch after the Easter break.

nanselm, 2023-04-05:
sounds wonderful. thank you and happy holidays.

LennertSchepers, 2023-04-25:
In my opinion tele**metry** involves a kind of measurement (doesn't require communication, data can also be stored on a device) so is different from tele**communication**. The description is only on the transfer of data so maybe *telecommunication systems* is better?

roy-lowry, 2023-06-19:
The description of this category has semantically drifted since I last looked.  I have concerns about the addition of protocols as this would dictate that 'HTTP' is a valid L22 entry, which makes me very uncomfortable. I would suggest a description along the lines of:

'Devices or systems that transmit data from one geographic location to another (e.g. WIFI, cellular phone or satellite networks)' 
'Telecommunication Systems' conveys a clear meaning to me and includes kit such as Iridium that are obviously within L05/L22 scope. I'm far less certain about the meaning of 'Telemetry Systems'.

danibodc, 2023-08-31:
Thanks for everyone's feedback and apologies for letting this drop off the radar. We have now loaded the following L05 category and it will be live on the NVS from tomorrow morning:

398	telecommunication systems
'Devices or systems that transmit data from one geographic location to another (e.g. WIFI, cellular phone or satellite networks)'

danibodc, 2023-09-01:
https://vocab.nerc.ac.uk/collection/L05/current/398/

closed 2023-09-01 by danibodc