nvs-vocabs/L05 #24: NTR: Non-dispersive infrared absorption (NDIR) carbon dioxide gas detector

lqjiang, 2023-04-26:
## Required
### Term name (PrefLabel)
[Non-dispersive infrared absorption (NDIR) carbon dioxide gas detector]
### Definition
[Nondispersive infrared (NDIR) spectroscopy is often used to detect gas and measure the concentration of carbon oxides (for example, carbon monoxide and carbon dioxide). An infrared beam passes through the sampling chamber, and each gas component in the sample absorbs some particular frequency of infrared.] 
### Sources/references
[[Please enter a link to a reference source](https://www.analog.com/en/analog-dialogue/articles/complete-gas-sensor-circuit-using-nondispersive-infrared.html#:~:text=Nondispersive%20infrared%20(NDIR)%20spectroscopy%20is,some%20particular%20frequency%20of%20infrared.)]

## Optional
### Synonym or acronym (AltLabel)
[Only use if the new device category has known synonyms or acronyms]
### Mapping to external terminologies
[Is your submission related to any existing terms in external resources: please enter one or a list of URIs]
### ORCID
[Enter your ORCID if desired]

lqjiang, 2023-04-27:
Would you please change the request to the below?

**Standard name:** DIC analyzers based on a CO2 gas detector
**Description:** DIC analyzers based on a CO2 gas detector including Non-dispersive infrared absorption (NDIR) (e.g., Licor LI-850), Cavity Enhanced Absorption Spectroscopy (e.g., Licor’s LI-7815), and  Cavity Ring-Down Spectroscopy (CRDS) (e.g., Picarro G2131i) detectors.

danibodc, 2023-07-27:
Hi @lqjiang, many thanks for your request.

We have a number of Li-Cor, Picarro, and other NDIR sensors in the L22 collection that are mapped to the following L05 categories:

dissolved gas sensors: http://vocab.nerc.ac.uk/collection/L05/current/351/
atmospheric gas analysers: http://vocab.nerc.ac.uk/collection/L05/current/382/

These would be suitable to use for these types of sensors. We would not want to create a more specific L05 category that describes the types of measurement e.g DIC in this case. Depending on the individual instruments in question, either of the above should hopefully be sufficient.

If you have any instruments that are not present in the main L22 collection, feel free to put in a request at https://github.com/nvs-vocabs/L22/issues.

closed 2023-07-27 by danibodc