nvs-vocabs/L05 #25: NTR: sample dividers

roswri, 2023-05-24:
## Required
### Term name (PrefLabel)
sample dividers

### Definition
A device used to subdivide material samples into two or more smaller portions.

### Sources/references
n/a

## Optional
### Synonym or acronym (AltLabel)
n/a

### Mapping to external terminologies
n/a
### ORCID
0000-0002-1711-6478

roy-lowry, 2023-05-31:
We used to call these 'sample splitters'. Might be worth including that in the description to provide a search target.

Osanna123, 2023-06-19:
I agree with Roy about adding 'sample splitters' as a synonym. 
Also, Iød suggest to edit the definition a bit to: 'A device used to subdivide material samples into two or more representative portions.'

roswri, 2023-06-27:
Thanks both for the suggestions, new code created:
397 - sample dividers (http://vocab.nerc.ac.uk/collection/L05/current/397/)
AltLabel=sample splitters	
Description: A device used to subdivide material samples into two or more representative portions.

closed 2023-06-27 by roswri