nvs-vocabs/L05 #27: NTR: Seine nets

gwemon, 2023-12-18:
From Neil Holdsworth by email:
I would advise referring to the FAO terminology as the accepted master reference for Fisheries technology: http://www.fao.org/faoterm/viewEntry/en/?entryId=98138 
Best, Neil

seine net 
Definition
Nets which are usually set from a boat, and can be operated either from the shore (beach seines) or from the boat itself (e.g., Danish or Scottish seines). The manner of capture is to surround an area of water with a very long net, with or without a bag at the centre. The net is usually operated by two ropes fixed to its ends, used both for hauling it in and for herding the fish. 
Definition source
Nedelec , C. and J. Prado (1990): Definitions and classification of fishing gear categories. FAO Fisheries Technical Paper, 222 (Rev. 1): 92 p. 
Related terms

From: [sdc-tech-request@listes.seadatanet.org](mailto:sdc-tech-request@listes.seadatanet.org) <[sdc-tech-request@listes.seadatanet.org](mailto:sdc-tech-request@listes.seadatanet.org)> On Behalf Of "Wright, Danielle E."
Sent: 15 December 2023 10:24
To: [sdc-tech@seadatanet.org](mailto:sdc-tech@seadatanet.org)
Subject: [sdc-tech] New L05 - seine nets

Hello,

We are proposing to create a new L05 term ‘seine nets’ with the following definition:

‘A surrounding net designed to sample fish that hangs vertically in the water with its bottom edge held down by weights and its top edge buoyed by floats. Seine nets can be deployed from the shore as a beach seine, or from a boat.’

gwemon, 2023-12-18:
@danibodc The FAO identifier for the term is: http://aims.fao.org/aos/agrovoc/c_73a64ed1 - It would be good to set up an exact match link between our new L05 concept and the FAO term.

danibodc, 2024-02-07:
Thanks, the mapping has been added, closing ticket.

closed 2024-02-07 by danibodc