nvs-vocabs/L05 #28: NTR: fyke net

danibodc, 2024-01-15:
We have received a request for a new L22 term: twin fyke nets (BODCNVS-1962).

A new L05 term 'fyke net' will be required. 

A definition has been found at https://www.fao.org/faoterm/viewEntry/en/?entryId=98431, as below:

Normally used in shallow water, consists of a cylindrical or cone-shaped bags mounted on rings or other rigid structures, completely covered by netting and completed by wings or leaders which guide the fish towards the opening of the bags. The fyke nets, fixed on the bottom by anchors, ballast or stakes, may be used separately or in groups.
Definition source: 
Nedelec , C. and J. Prado (1990): Definitions and classification of fishing gear categories. FAO Fisheries Technical Paper, 222 (Rev. 1): 92 p.

However, 'fyke net' is only listed as a synonym to 'trap nets' in the Agroportal: http://aims.fao.org/aos/agrovoc/c_eac2ecd2

danibodc, 2024-01-24:
Associated with https://github.com/nvs-vocabs/L22/issues/225

danibodc, 2024-01-31:
Proposing to create a more generic L05 term for 'trap nets' with definition:

Stationary nets such as Fyke nets that are used in shallow waters such as coastlines or estuaries. They are fixed to the bottom by anchors, ballast or stakes, and may be used separately or in groups. They often consist of a labyrinth-like chamber into which fish can easily enter, and from which they cannot easily escape. Target species include salmon, trout and eel.

gwemon, 2024-01-31:
@danibodc we do have a term for passive fishing gears: https://vocab.nerc.ac.uk/collection/L05/current/63/ I wonder were you thinking of having a broad/narrow relation of the new term with that one?

danibodc, 2024-01-31:
Hi @gwemon yes I think a mapping to https://vocab.nerc.ac.uk/collection/L05/current/63/ is a good idea, agree it is quite boad.

neil-ices-dk, 2024-01-31:
This logic follows the EU specific Metier definitions, level 3 refers to 'Traps' https://vocab.ices.dk/?CodeID=169686
and level 4 (more detailed) refers to 'Fyke nets' https://vocab.ices.dk/?codeguid=68397d52-d86a-4a6c-ad36-1012c0d7c68a

danibodc, 2024-02-02:
The term has now been loaded and is live on the NVS:
https://vocab.nerc.ac.uk/collection/L05/current/74/

closed 2024-02-02 by danibodc