nvs-vocabs/L05 #32: NTR: Request for L05 Remotely operated vehicle

nvsvocabs, 2024-10-29:
<H2> Required </H2><p><b>Term name (PrefLabel)</b><p>Remotely operated vehicle <p><b>Definition</b><p> ROVs are unoccupied, highly maneuverable underwater machines that can be used to explore ocean depths while being operated by someone at the water surface. Most ROVs are equipped with at least a still camera, video camera, and lights, meaning that they can transit images and video back to the ship. Additional equipment, such as a manipulator or cutting arm, water samplers, and instruments that measure parameters like water clarity and temperature, may also be added to vehicles to allow for sample collection.<p><H2> Optional </H2><p><b>Sources/references</b><p> https://oceanexplorer.noaa.gov/facts/rov.html<p> <b>ORCID</b><p> https://orcid.org/0000-0003-0974-3401<p> <b>Synonym or acronym (AltLabel)</b><p> ROVs<p>

rubenpp7, 2024-10-29:
Forgot to add a similar vocab term to the request, but it exist:
https://mmisw.org/ont/ioos/platform/remotely_operated_vehicle

JordanAthertonNOC, 2024-11-01:
Hi @rubenpp7 

This seems more appropriate for the [L06 ](https://vocab.nerc.ac.uk/collection/L06/current/)(SeaVoX Platform Categories) collection rather than the [L05 ](https://vocab.nerc.ac.uk/collection/L05/current/)(SeaVox Instrument Categories) collection.

rubenpp7, 2024-11-05:
Hi @JordanAthertonNOC ,
I understand how that makes sense, I added it here because our users name "ROVs" as the instrument that took the videos/samples

JordanAthertonNOC, 2024-11-11:
Apologies I should've also checked at the same time whether we already had something in L06 that fit the bill. We do have the following existing entry which references ROVs in the term description, does this fit your purpose? 

ID: `[22](https://vocab.nerc.ac.uk/collection/L06/current/22/)`
Preflabel: `propelled unmanned submersible`
Description: `A platform operating in the water column attached to a mothership by an umbilical with limited propulsion and no human operator on board (e.g. ROV).`